jessecusack/perturb #1: Guiding principles and first steps

jessecusack, 2023-08-03:
Following our meeting this morning, here are our goals for this package:

* Produce the highest quality possible estimates of epsilon and chi (with error estimates and georeferencing)
* Be very well documented and transparent (lots of inline comments)
* Be easy to automate (fit into ship based operations where new data comes in bit by bit) and configurable using parameters
* Following the spirit of ATOMIX standards, e.g. L1, L2, L3, L4 processing and outputting standardized data that is CF compliant

Important, but secondary goals:
* BBL focused processing
* Handling of DO/Ch/Turb data

Things to think about down the road
* Calculating chi. FP07 calibration drift?

mousebrains, 2023-08-04:
I'm slowly switching to snake case, and we can migrate naming conventions to be roughly in agreement with the "community guidelines" Jesse linked to.

For a prefix, I have been using osgl_, Oregon State Glider Lab, for my other public packages. We can use that, or come up with something else. But yes, it would be a good idea for code isolation.

jessecusack, 2023-08-07:
I don't like osgl, but happy to brainstorm other options.