jessecusack/perturb #2: Implement core functionality with unit tests

jessecusack, 2023-08-03:
Develop the code that gets us from p files to dissipation estimates per profile. This is actually quite a big task with lots of substeps that may need to be broken down. The main parts are:

- [ ] L1 processing: extract timeseries from p files. This is essentially a loop over odas p2mat. 
- [ ] L2 processing: perform basic QC, such as despiking and separate the timeseries into profiles. 
- [ ] L3 - 4 processing: estimate spectra and calculate turbulence quantities on a per profile basis.

mousebrains, 2023-08-04:
A working version of my code is now checked in. There is a script to run it in Examples/example.m using the dataset that is in repository.

L1 is process, but needs work for the naming conventions, QC, and unit tests.
L2 is processed, but needs work for the naming conventions, QC, and unit tests.
L3 is processed, but needs work for the naming conventions, QC, and unit tests.

I believe I've made all the public facing naming convention consistent with snake, but there are still a lot of internal variables that are in camel.

jessecusack, 2023-08-07:
Great, thanks Pat. I'll try running it later this week.