jessecusack/perturb #3: Documentation

jessecusack, 2023-08-04:
@mousebrains, how do we want to deal with documentation? 

It _might_ be possible to use sphinx/readthedocs, e.g. https://github.com/sphinx-contrib/matlabdomain. Maybe this is too complicated for our needs and asking for trouble down the line?

One simple option is to write a very extensive README file in markdown. I like this because it has no dependencies, but it does require more effort to keep up to date.

mousebrains, 2023-08-04:
I'll dig through a method that can use the GitHub workflows. I don't know of one for Matlab currently.

I would love to have the following sources of documentation:

- External high level documentation, think overview
- Internal to the code documentation ranging from high level to micro details
- Supplementary documentation, like papers

jessecusack, 2023-08-07:
If you find a suitable solution with workflows we can discuss here. 

I think we should aim for a solution that doesn't add too much overhead/dependencies for the project and is simple to modify for anyone who is less proficient with git. To be honest, I would be happy to just have a very extensive README.