jessecusack/perturb #10: Simplify GPS

jessecusack, 2023-08-22:
I think that the GPS file parsing functionality is beyond the scope of this package. We should assume that the user has the capability of creating a lat/lon time series to feed into our processing. It should be completely optional to specify GPS information at all. I can imagine a whole package dedicated to creating nicely formatting GPS files from NMEA strings and other formats.

mousebrains, 2023-08-23:
Am I missing something? There is nothing in this package about how to create the GPS file, rather a few methods, including a no-GPS method, to suck in GPS data created external to the package.

One of my external packages for the Thompson sucks in the NEMA GPS and INS sentences, then corrects the ship's GPS fixes to an approximate GPS fix for where the VMP was being dropped from off the fantail. The output of the external package is used as input to the VMP processing package.

jessecusack, 2023-08-23:
The methods that suck in GPS data requires that a file is present with the variables (or headers) `t`, `lon`, `lat`. So the user has to first create a file in this format before they can use the package. I am ok with having some generic GPS file reading capability. At zero order, the package should support supplying time, lon and lat as arrays.

mousebrains, 2023-08-23:
I've had to deal with too many ships with different methods of obtaining the GPS data, so I wanted a well defined API, GPS_base_class.m, for this package to deal with, but localized user extensible facilities that would initialize GPS_base_class.m using the time, lat, and lon data sources.

The GPS_base_class already deals with the GPS data the way you're envisioning it, but let me think about how the GPS classes are initialized in mat2profile, which uses the "filename."  I don't want to load the underlying GPS data unless it is needed.

My off the top of my head method would be for the user to instantiate a class derived from GPS_base_class, and then to have mat2profile call a method, initialize, which loads the data, if necessary. In your use case, the user would instantiate the GPS_base_class directly with the time, lat, and lon variables.

mousebrains, 2023-08-23:
@jessecusack Please take a look at the GPS initialization pull request and see if this satisfies what you're thinking.

mousebrains, 2023-08-29:
I've closed the GPS pull request so I can move onto the next task on my list. I also added a new method GPS_from_vectors per a request from Jesse.

mousebrains, 2023-09-04:
Added documentation about the GPS classes and examples for how to use them.

mousebrains, 2023-09-04:
@jessecusack can we close this issue?

mousebrains, 2023-10-03:
Nothing back from Jesse, so I'm closing this.

closed 2023-10-03 by mousebrains

jessecusack, 2023-10-03:
I'll leave it closed for now but may reopen at a future time.