jessecusack/perturb #5: Use of temporary settings for save file version

jessecusack, 2023-08-22:
I don't think that we should be editing global matlab settings like this:

https://github.com/jessecusack/perturb/blob/0a4928fb4df6f3e3354a924ade6393fa0a3154b8/Code/process_VMP_files.m#L12-L14

My understanding is that this will change user settings until they quit and reopen matlab. This could lead to unexpected issues for users that run our scripts and then do something else without closing matlab first. I would suggest that we create an optional parameter for save file type and pass it to the save function directly.

mousebrains, 2023-08-22:
Yes, you are correct, it would change things for the session. There is now a matlab_file_format parameter which is used in all the save commands.

mousebrains, 2023-08-22:
Checked in

jessecusack, 2023-08-22:
Reopening because I see no commits that address this.

mousebrains, 2023-08-23:
I had committed the changes but forgot to push them. They are now pushed.

jessecusack, 2023-08-23:
Solved by c4048a48e7d84e26aad0434fb73af9b2509eb668

closed 2023-08-23 by jessecusack