jessecusack/perturb #6: Miscellaneous small issues

jessecusack, 2023-08-22:
I am currently reading through the code and will add to this issue with small things as I find them.

Use of `true` here forces print. Make optional via a parameter?
https://github.com/jessecusack/perturb/blob/0a4928fb4df6f3e3354a924ade6393fa0a3154b8/Code/process_VMP_files.m#L20

mousebrains, 2023-08-22:
Changed to using the already existing debug parameter supplied by get_info or the user.

jessecusack, 2023-08-22:
The point of this issue is to compile miscellaneous small issues rather than creating a new issue every time. I am still reviewing the code and may find more. Reopening.

jessecusack, 2023-08-22:
Employ more deliberate printing method?

https://github.com/jessecusack/perturb/blob/b69ed0e942da2ee86d26ba2edc9e25579a1db4ea/Code/mk_filenames.m#L39-L40

mousebrains, 2023-08-23:
This was left over from debugging. It has been removed.