jessecusack/perturb #8: Netcdf formatting

jessecusack, 2023-08-22:
The formatting of the long names and units needs simplifying. Keywords are overly broad. 

I see the long_name as the name I would want displayed as an axis label. The units should be as concise as possible, within the CF/udunits convections, and standardized across variables. 

Examples of long names to fix:
- pressure_slow -> Pressure
- stime -> Time

Examples of units to fix:
- Watts -> W
- decibar -> dbar
- meter -> m
- seconds / sec -> s
- Volts -> V

Not all of the following keywords will appear in all datasets, I suggest we remove this attribute. 
https://github.com/jessecusack/perturb/blob/b69ed0e942da2ee86d26ba2edc9e25579a1db4ea/Code/Combo.json#L390-L391

Files:
https://github.com/jessecusack/perturb/blob/main/Code/Combo.json
https://github.com/jessecusack/perturb/blob/main/Code/CTD.json

mousebrains, 2023-08-23:
Please go ahead and change the JSON to what you would like. As long as the output passes CF-1.8 and ACDD-1.3 compliance checkers I'm good.

Much of the long units names were done for ACDD-1.3 compliance.

jessecusack, 2023-10-17:
Mostly fixed by #14

closed 2023-10-17 by jessecusack