jessecusack/perturb #9: Break down package entry points (and processing steps)

jessecusack, 2023-08-22:
The main entry point this package for end users is the function `process_VMP_files`, which wraps all the processing steps together. There are many occasions where someone might want to process a single p file (or even a single profile), without creating a whole new folder structure. I think we need to unwrap the processing steps and make them accessible to users.

mousebrains, 2023-08-23:
You can process a single P file by modifying the p_file_pattern.

Single profiles will break some of the algorithms, like top trimming, FP07 calibration, and CTD/DO/Fluorometer/Turbidity which use an entire P file.

We can construct the processing of a P file into a pipeline:

1. odas_p2mat the .P file to a .MAT file
2. Split up the profiles and full file scalar information
3. Bin up the profiles and full file scalar information

Which is then wrapped by the macro handling of all the P files as a whole.

From a development and unit testing standpoint I don't prefer this method.

alesss42, 2023-08-23:
Hi! Thank you for writing all this. 

I would be interested in processing a single P file but also just a section; for example, I want to look into details of the measurements of a subduction event from SUNRISE. Would creating a folder just with the P files I'm interested in be a good idea? And that folder be "fed" to the `process_VMP_files`.

mousebrains, 2023-08-23:
@alesss42 That is how I would initially think about processing a section, either by how the filenames are name or in a directory. You can use symbolic links if you want to save space.

mousebrains, 2023-10-19:
Cleaned up in refactorizatioin

closed 2023-10-19 by mousebrains

jessecusack, 2023-10-23:
If you link an issue to that pull request then it will automatically close when the pull request is merged into the main branch.