jessecusack/perturb #16: create noise flag instead of dropping epsilon

jessecusack, 2023-10-03:
I think that the appropriate way to handle epsilon values close to the instrument noise floor is to create a true/false QC variable, not to convert estimates to NaN. This would be the transparent way to handle QC and allows for the adding of additional QC flags in the future. 

The parameter `diss_epsilon_minimum` should be renamed to something like `epsilon_noise_floor`.

mousebrains, 2023-10-03:
diss_epsilon_minimum was not put in for the oceanographic operational noise floor, but to solve an issue with broken electronics in the VMP. The levels we were seeing were 1-3 orders below the instrument's operational noise floor. 

This being said, I'm good with changing to a QC method. I'll look at a clean method of implementing QC flags.

Since this is not to go after the operational noise floor, I'll go with a different name than `epsilon_noise_floor` and `diss_epsilon_minimum`, but I understand your logic.

mousebrains, 2023-10-31:
#22 Changed default from 3x10^-10 which was appropriate for bad electronics to 1e-13, which should "never" be hit with "normal" electronics.

closed 2023-10-31 by mousebrains