jessecusack/perturb #17: Make parallel processing optional

jessecusack, 2023-10-05:
@mousebrains, what do you think about adding an optional flag for parallel processing and throwing in a parfor loop somewhere? I know that some users may not have the right toolbox, but for those that do, it could speed things up immensely. I believe that batch processing of independent p / mat files qualifies as an 'embarrassingly' parallel problem.

mousebrains, 2023-10-05:
I've thought about this, but has not risen to the top of my priority list yet. In the past, 5+ years ago, I've had problems with Matlab's inherently non-threaded design causing issues. That being said, yes parallel processing from P files through the creation of the profiles is a logical thread.

jessecusack, 2023-10-05:
The nearly 40 to 50 second run time seems way too long to me. Is this what you get too?

```
A685_0152 spliting into 1 profiles
WARNING: A685_0152 profile 1 GPS fix 873812:42:59 from t0 12-Jun-2023 05:13:39
A685_0152 took 49.41 seconds to extract 1 profiles
A685_0153 spliting into 1 profiles
WARNING: A685_0153 profile 1 GPS fix 873815:44:32 from t0 12-Jun-2023 08:15:12
A685_0153 took 49.00 seconds to extract 1 profiles
A685_0154 spliting into 1 profiles
WARNING: A685_0154 profile 1 GPS fix 873818:44:04 from t0 12-Jun-2023 11:14:44
A685_0154 took 48.79 seconds to extract 1 profiles
A685_0155 spliting into 1 profiles
WARNING: A685_0155 profile 1 GPS fix 873821:46:08 from t0 12-Jun-2023 14:16:49
WARNING: No trimable casts found in A685_0155
A685_0155 took 1.37 seconds to extract 1 profiles
A685_0156 spliting into 1 profiles
WARNING: A685_0156 profile 1 GPS fix 873821:48:53 from t0 12-Jun-2023 14:19:33
WARNING: No trimable casts found in A685_0156
A685_0156 took 1.99 seconds to extract 1 profiles
A685_0157 spliting into 1 profiles
WARNING: A685_0157 profile 1 GPS fix 873821:54:51 from t0 12-Jun-2023 14:25:31
WARNING: A685_0157 cast 1 dropping 16.11% epsilons outside of 95% confidence interval, iter=1
WARNING: A685_0157 cast 1 dropping 12.85% epsilons outside of 95% confidence interval, iter=1
A685_0157 took 40.69 seconds to extract 1 profiles
```

At this rate it will take 4 - 6 hours to processes the whole dataset.

mousebrains, 2023-10-06:
That's interesting, on my M1 MacBook Pro, I chomped through all the MR files for both gliders in ~1.5 hours in Saipan. Still longer than I want, but significantly less than you're seeing. That was with Matlab 2023a, the x86 version. I'm now running 2023b, the ARM version, and I'm getting ~20-40% speed improvement on my tests so far.

jessecusack, 2023-10-06:
Turns out I had 2023b on my laptop and it is twice as fast now. 

I still think it could be faster with parfor!

mousebrains, 2023-10-19:
Done in the refactoring and parallelization. See the Pull Request.

closed 2023-10-31 by mousebrains