jessecusack/perturb #18: Random string in folder name

jessecusack, 2023-10-05:
@mousebrains can you remind me why we append a random string to some of the output filenames?

mousebrains, 2023-10-05:
It is not a random string, it is a hash of the configuration parameters.

One of the issues I've encountered is people change the configuration parameters, without wiping out all the data processed with the previous set of configuration parameters. Using the hash eliminates this problem.

I looked at more "human understandable" options, but I didn't find any. I don't remember if I've done this yet, but my plan had been to create a JSON/text version of the configuration options that go into the signature within the directory, so a human can understand how the directory was created.

This only becomes an issue if one allows for marginal analysis. i.e. if one adds P files, then one only needs to process the new stuff. This is the typical mode on ship.

jessecusack, 2023-10-05:
The JSON/text solution is the right way to go because, like you say, it is human readable. I would do this an append a number like 001, 002 etc. of needed.

mousebrains, 2023-10-05:
Let me think about the number versus hash. The hash is "easy," in that I don't have to look in each directory with a number to figure out if it is the correct directory.

jessecusack, 2023-10-05:
Can you store the hash within the directory?

If only matlab supported YAML, all this config stuff would be a lot easier.

mousebrains, 2023-10-06:
The hash would definitely be there, in case the user "hand edits" the JSON/text. I'll have to explicitly check that the hash of the JSON/text matches the version I've stored.

Having multiple directories is not a big deal, as long as the number is in the handful range, not thousands.

jessecusack, 2023-10-06:
I opened a second matlab instance and ran the same parameters yet it created a new profiles folder. I don't understand how this is working...

mousebrains, 2023-10-19:
Changed in upcoming refactoring and parallelization

jessecusack, 2023-10-23:
Great, linked to PR.

closed 2023-10-31 by mousebrains