jessecusack/perturb #19: saving shear spectra uses too much storage

jessecusack, 2023-10-06:
For a single 1000 m depth profile we go from a ~30 MB p file, to a ~110 MB p2mat file to a ~ 500 MB profile file... a whopping order of magnitude increase in storage requirements. 0.5 GB per profile is too much. I think that saving the p2mat file makes a lot of sense because it saves time loading the data, but we should not save all windowed shear / acceleration spectra.

jessecusack, 2023-10-06:
What is the bbl structure? Why does it exist?

mousebrains, 2023-10-09:
BBL is a bottom up dissipation estimate. It came from conversations with Fucent and you thinking about bottom crashing. The concept was to add bottom crash detection, then to estimate the epsilons near the bottom better, possibly with different parameters than the top down.