jessecusack/perturb #20: Epsilon estimated from separate shear probes in binned data?

jessecusack, 2023-10-10:
It seems that the binned data (and therefore the combo netcdf) does not contain separate estimates of epsilon from the two shear probes. I haven't figured out where they get dropped, but they should be outputted. How would we change this?

jessecusack, 2023-10-30:
@mousebrains I think this is happening in the netcdf creation step. My current guess is because `e_diss` is a 2 x n matrix, only the first column is being saved so really we're only seeing shear probe 1 data.

mousebrains, 2023-10-31:
#22 Exposed individual shear probe data

mousebrains, 2023-10-31:
Here is an example for a VMP cast from the combo file:
[image]

closed 2023-10-31 by mousebrains