jessecusack/perturb #21: Trimming surface data?

jessecusack, 2023-10-10:
When I run `quick_look` on some of our data, I can get epsilon estimates to within 2 m of the surface. From looking at the combo file, it looks like surface data is being dropped by our processing. Where in the code do you think this might be happening?

mousebrains, 2023-10-19:
In my results, I get data up to 1m.

jessecusack, 2023-10-20:
Can you copy paste your configuration file here?

jessecusack, 2023-10-20:
This plot compares the output of `quick_look` using `ql_info.profile_min_W = 0.1;` with the output of the perturb combo file using the same vertical velocity threshold. `quick_look` produces epsilon to shallower depths. peturb seems to be biased low compared to quick_look, sometimes by an order of magnitude.

[image]

jessecusack, 2023-10-23:
This issue was solved by setting the parameter `trim_use = false`

closed 2023-10-23 by jessecusack