jessecusack/perturb #25: add original filename to combo output

jessecusack, 2023-10-31:
It would be super helpful to add a variable to the combined output that contained the original data file name. That would make it super easy to go back and investigate suspect profiles in detail. If it could also include the profile number for extra utility.

mousebrains, 2023-10-31:
In the combo info table, the name column allows you to understand the files that lead to that profile. The index column is the profile number within the profiles.

jessecusack, 2023-10-31:
I guess I specifically meant the netCDF output.

mousebrains, 2023-11-01:
Fixed in a patch I just pushed to the main branch. I had explicitly dropped them since one can't compress strings in Matlab's NetCDF interface. I now have a check if the variable is a string, in which case it doesn't tell NetCDF to compress the variable.

mousebrains, 2023-11-01:
In the previous version I had the absolute filenames. I took that away in this version and just use the filename, sans suffix, as an overall key through the various tables. This was to cleanly handle shipboard use where multiple people will be running marginal updates and their paths may be different on their individual systems.

closed 2023-12-09 by mousebrains