jessecusack/perturb #32: diss inspector modifications

jessecusack, 2024-01-07:
@mousebrains the diss_inspector is incredibly useful, thank you for making it! After playing around a little, I have some suggestions for rearranging the plots. In the figure with diss plots, instead of viscosity and temperature, it would be helpful to plot shear, acceleration, pitch battery position, and buoyancy engine changes. This way it would be easier to see if a shear signal might be attributed to real ocean turbulence or some internal glider movement. Additionally, a plot of U_EM on top the of the speed would be helpful. I noticed that U_EM (when it seems reasonable) is usually higher than speed by about 10 - 20%.

mousebrains, 2024-01-08:
@jessecusack We'll have to modify the hotel file generated from the DBD/EBD files, then reprocess everything from the P files forward. Once that is done adding glider related measurements should be straightforward.

For shear, are you thinking from the EM sensor? Vertical or longitudinal? 

I'm sure it will take a few iterations of what to include in the hotel file to get to where you're thinking.

It will probably be 2+ weeks before I get to it.

mousebrains, 2024-01-08:
In theory, the buoyancy engine and battery position were fixed when the MR was one. The thruster should only have come one in the surfacing behavior if the glider slowed down too much. (I have not explicitly checked all these are true yet.) So the only mechanical noise would come from the tail fin.

mousebrains, 2024-01-28:
The data is presented pre and post dissipation calculation. The pre is the profile data and is continuous in time. The post is chunked up and averaged according to the FFT/Diss lengths. The shear spectra is in the profile data set, so I don't really want to move it to the dissipation plots.

mousebrains, 2024-01-28:
When I get the hotel file generation updated with the motor movement information, I'll figure out how to allow the user to add supplemental plots for the profiles.