jessecusack/perturb #36: serial number in combo files

Mareike-Koerner, 2024-08-07:
Is there a way to add the serial number of the instruments to the final combo files? For the sunrise data there is one VMP which has an offset in the salinity sensor. Working with just the final combo files is is not possible to extract the affected profiles. Would it be possible to include the serial number of the instrument as a variable in both the dissipation combo and the ctd combo files?

mousebrains, 2024-08-07:
Hi Mareike,

Yes it should be, provided I can extract it from the config information.
I'm working on code to calculate Chi now, so I'll look at adding the serial
number as part of that effort.

Pat

mousebrains, 2024-08-08:
The serial number has been added as part of the ChiHooks branch. #34

PR #38 (2025-02-01): take parts of ChiHooks branch that impliment SN 
Closes #36
closed 2025-02-01 by jessecusack