jessecusack/perturb #41: Simplest binned science output

jessecusack, 2025-02-12:
Unless I am mistaken we don't currently compile the simplest set of binned scientific variables into a single file. This is typically what is needed for making plots and performing further analysis. 

That output file should contain:
* Epsilon 1 & 2 as a function of depth / z
* Temperature
* Practical salinity
* Potential density referenced to 0 dbar
* Conservative and absolute salinity
* Buoyancy frequency
* Vehicle speed/U_EM if it exists
* Pressure
* Lat, lon

And nothing else (?). 

I would prefer not to create another output directory if possible.

mousebrains, 2025-02-12:
It is actually all there, but split into two combo files for two reasons:

- The depth binning is different between scalar and dissipation estimates. One can bin the scalar variables much finer and get good statistical estimates, while the dissipation estimates need a courser binning.
- There is a lot of ancillary data attached to the dissipation estimates for determining if one should believe the dissipation estimates.

If one is willing to give up the binning difference, they can be easily bin the data into equal sized bins and extract just what you're looking for.

The third combo file is binned in time and is only the scalars. It is not split into profiles.

jessecusack, 2025-02-13:
We rarely use the higher resolution CTD data in isolation. For analysis purposes we almost always want to combine it into a common data structure with microstructure, shipboard ADCP, etc and will generally be limited to the resolution of some other sensor.

jessecusack, 2025-02-13:
@mousebrains what is the time binned scalar data used for?

mousebrains, 2025-02-13:
I use the fine binned scalar data all the time, in addition to the coarser binned dissipation data. When I'm looking for fine layer structure and correlations between the various scalar sensors, temperature, conductivity, DO, chlorophyll, backscatter

The time binned data includes the upcast scalar data, which helps fill in the horizontal gaps. I have a model of how the VMP comes up which is then translated into an "effective" lat/lon. For some scalars, the data on the upcast has caveats, fast temperatures, conductivity, but for others, JAC-T, backscatter, chlorophyll, DO, it is fine. Even the fast T and conductivity show reasonable agreement with the downcast data.

mousebrains, 2025-05-02:
@jessecusack Are you good with closing this issue?

jessecusack, 2025-05-08:
Not yet. I still want perturb to be able to produce some output with turbulence variables and CTD variables combined.