jessecusack/perturb #42: conductivity sensor

jessecusack, 2025-02-14:
Add a parameter to specify a constant offset to the conductivity measurement to fix calibration errors.

mousebrains, 2025-02-14:
Are you thinking of just an offset or also a slope?

jessecusack, 2025-02-15:
Just a constant offset for now.

mousebrains, 2025-02-16:
Any logical objections to tucking this into the convert2mat phase from P files. This way the change will propagate to all the downstream uses of conductivity. But if one changes it, then one will regenerate everything, including the odas_p2mat processing.

With just a constant offset, I don't the odas computation of gradC will be impacted. This is done as part of the odas_p2mat processing.

jessecusack, 2025-02-18:
convert2mat seems like the best place for it.

mousebrains, 2025-04-09:
@jessecusack In thinking about this more, does it make sense to patch the underlying P file sensor coefficients for conductivity? Then it will propagate through the system  cleanly and be patched for all future uses of the P file?

jessecusack, 2025-04-18:
That is a good point @mousebrains. I think that is the correct approach. It makes me wonder if perhaps the configuration file for SN192 was accidentally edited and the offset was introduced by accident. Do you know how we can dig up the default configuration from the VMP?

mousebrains, 2025-04-18:
This is the second VMP I've been involved with that has had a problem with the JAC-C calibration in the last year.

It could be a problem with the calibration coefficients in the config or it could be the calibration has shifted. 

Rockland's recommendation is the JAC sensor be recalibrated every year due to drift. This involves sending it back to Rockland and the base cost I just learned will be ~3k.

For Sophia's VMP, which also has an issue, I have ~48 hours of comparison against 4 other CTDs, so I'm able to do a full multi-point calibration.